kbembedded/Flipper-Zero-Game-Boy-Pokemon-Trading #2: Flipper not detected or connection closed

FoxLy1312, 2023-03-26:
Instructions probably unclear 

trying to connect the pins directly to a cutted Link-Kabel. 

my question is : where the resistor supposed to get soldered to ? 

[image]

EstebanFuentealba, 2023-03-26:
Hi, I used 33k on CLK, but it's optional, it can be connected directly.

EstebanFuentealba, 2023-03-26:
When you have the cable cut, test continuity with a voltmeter on the connector that goes to the Game Boy. It's a crossover cable, so if on the Game Boy it's SI SO, on the Flipper Zero it should be SO SI.

[image]

EstebanFuentealba, 2023-03-26:
@FoxLy1312  Confirm if it helped you to close the issue or if not, to review what could be done. If it helped you, please share the images with your GBA 👍.

FoxLy1312, 2023-03-26:
Combine all the suggestions dos the trick, thanks for the help, great work ♥️

[image]

closed 2023-03-26 by FoxLy1312