kbembedded/Flipper-Zero-Game-Boy-Pokemon-Trading #7: Traded Pokémon don't have names

LuxStatera, 2023-04-22:
I am using Unleashed and a Game Boy Color with Pokémon Yellow. I traded a Mew and it was successful but when I check my Pokémon the Mew has no name. It is just left blank

kbembedded, 2023-05-22:
@LuxStatera Take a look at my fork, I've modified `pokemon_data_h` to turn it in to a struct and make it a bit easier to modify. Give that a look and let me know if you have any questions.

LuxStatera, 2023-05-22:
@kbembedded Thank you! I did a little more testing with a buddy of mine to see if we could adjust the levels, moves, etc but it didn't work. Although we did manage to accidentally trade a missingno. Once we do more tests I will reach out to you with our findings and any questions that pop up

LuxStatera, 2023-05-22:
@kbembedded btw on line 101 you have #define 0_  0xff but it should be 9_ right? Also I tried to generate the fap file but it ends in an error. Lines 92 - 101 "macro names must be identifiers

kbembedded, 2023-05-22:
> @kbembedded btw on line 101 you have #define 0_ 0xff but it should be 9_ right? Also I tried to generate the fap file but it ends in an error. Lines 92 - 101 "macro names must be identifiers

Oops. I added the numbers after I proved the struct setup functional. Forgot macros can't start with a number. Pushed a change that just makes them `_#_`  as well as changed the second 0 to a 9. Thanks.

kbembedded, 2023-05-22:
Oh. Something else you should probably know. If you want to trade pokemon to yourself that your game thinks you caught, I _think_ all that you need to do is set the OT Name to your name in the game and then the `.orig_trainer` struct member to match your in-game trainer ID. I think later games added a "secret" ID that is also compared against, but I think in Gen I its just those two that need to match in order for the game to think you caught a pokemon.

LuxStatera, 2023-05-22:
Happy to help. I don't know much about this stuff but I am learning. I was able to generate the fap file and the Pokemon's name changed but the name was "Flopper". I saw that the traded Pokemon needs the trainer ID and secret ID but I am not sure where to find my secret trainer ID. I'm not sure if this is possible but maybe the Pokemon we trade from the Game Boy that has our trainer ID and secret ID can be copied and applied to the Pokemon traded from the Flipper. That way everyone won't have to edit the file with their individual trainer ID

kbembedded, 2023-05-22:
> Happy to help. I don't know much about this stuff but I am learning. 

Let me know if you have any questions. Much of this is going to be an undocumented mess as I make changes bit by bit to make integrating a in-Flipper menu system eaiser.

> I was able to generate the fap file and the Pokemon's name changed but the name was "Flopper". 

Yeah, the struct was changed to name the OT "Flipper" with an ID of `01234` (I think), and then each pokemon is nicknamed "Flopper." I named it differently so its more obvious in the struct. You only need to change the first instance of these in the struct to change these details as the struct is data on the whole party, but the party only has a single pokemon in it. The rest of the data is ignored by the game for that reason.

> I saw that the traded Pokemon needs the trainer ID and secret ID but I am not sure where to find my secret trainer ID. 

I'm not sure if this is possible but maybe the Pokemon we trade from the Game Boy that has our trainer ID and secret ID can be copied and applied to the Pokemon traded from the Flipper. That way everyone won't have to edit the file with their individual trainer ID

To be clear, there is no secret ID in Gen I, maybe Gen II, I'm not sure I havn't looked that far. So you should just need to match the `.ot_name` to your in-game name, and then the ".orig_trainer" value to match your ID (note that the numbers are backwards due to endianness differences; you want to convert your number to a 16bit hex, and then swap the hex pairs).

LuxStatera, 2023-05-22:
I have never used hex before so I don't know how to get the correct value for my trainer ID. My trainer ID is 02626 or 2626.

kbembedded, 2023-05-22:
> I have never used hex before so I don't know how to get the correct value for my trainer ID. My trainer ID is 02626 or 2626.

Here is a quick way to convert:
https://duckduckgo.com/?t=ffab&q=2626+to+hex&ia=answer  == `0x0a42` Endian swapped is `0x420a` Give that a try as your `.orig_trainer` value. If that doesn't work, then you'll have to wait till I get around to playing with that a bit more.

kbembedded, 2023-05-22:
> That worked :) My new Mew listened to my commands!

Cool!

> It started at level 74 then after gaining EXP it "leveled up" to 54

Yeah, the party data structure contains a lot of repeated data in Gen I. Likely just to save on run-time computation. e.g. leveling up is determined solely by total EXP and the current level has zero impact on this process, but is likely there so every time you pause the game doesn't have to re-calculate the current pokemon level for each one in your party.

The trade data structure is literally the same as the in-game party structure, again, likely to save on compute time when sending thing around.

R4g3D, 2023-09-11:
This should be fixed in the current code. The traded Pokemon should have no nickname (i.e. it's name should be the name of the Pokemon traded and change once evolved). It should be easy from this point to add a feature to pick the nickname too, I may work on that in thee future.
The "level up" but still remains unfortunately, not sure what's causing the bug since the traded Pokemon appears to have the level selected and the appropriate EXP. This is also some work to be done to debug the issue.
@EstebanFuentealba I think you can close this issue since the orginal bug should be fixed.

closed 2023-09-27 by EstebanFuentealba