kbembedded/Flipper-Zero-Game-Boy-Pokemon-Trading #8: Adjust Pokémon level before trade?

LuxStatera, 2023-04-22:
I noticed all the traded Pokémon have maxed out levels. This is awesome but unfortunately makes them unusable if you are starting the game over again. Pokémon won't listen to you if their level is too high and you don't have enough badges. Is there a way to edit their level before trading them?

EstebanFuentealba, 2023-04-23:
Yes, it is possible to edit all of the data for a Pokémon (level, attacks, names, etc.) by editing the "pokemon_data.h" file. I have provided the hexadecimal values for each data field that needs to be edited. Additionally, you can refer to https://bulbapedia.bulbagarden.net/wiki/Pok%C3%A9mon_data_structure_(Generation_I) for more information on the data structure.
https://github.com/EstebanFuentealba/Flipper-Zero-Game-Boy-Pokemon-Trading/blob/04e0332ff4b2dbd448232238102774a38445c5ce/pokemon_data.h#L14-L20

For example, the default level for a Pokémon is `74`. When converted to hexadecimal, it becomes `0x4A`. This is where you can make changes and experiment.

You can use a decimal to hexadecimal converter like https://www.rapidtables.com/convert/number/decimal-to-hex.html to help you with the conversion.

EstebanFuentealba, 2023-04-23:
To edit names you can refer to this dictionary.
```bash
' ' = "0x7F",
'A' = "0x80",
'B' = "0x81",
'C' = "0x82",
'D' = "0x83",
'E' = "0x84",
'F' = "0x85",
'G' = "0x86",
'H' = "0x87",
'I' = "0x88",
'J' = "0x89",
'K' = "0x8A",
'L' = "0x8B",
'M' = "0x8C",
'N' = "0x8D",
'O' = "0x8E",
'P' = "0x8F",
'Q' = "0x90",
'R' = "0x91",
'S' = "0x92",
'T' = "0x93",
'U' = "0x94",
'V' = "0x95",
'W' = "0x96",
'X' = "0x97",
'Y' = "0x98",
'Z' = "0x99",
'a' = "0xA0",
'b' = "0xA1",
'c' = "0xA2",
'd' = "0xA3",
'e' = "0xA4",
'f' = "0xA5",
'g' = "0xA6",
'h' = "0xA7",
'i' = "0xA8",
'j' = "0xA9",
'k' = "0xAA",
'l' = "0xAB",
'm' = "0xAC",
'n' = "0xAD",
'o' = "0xAE",
'p' = "0xAF",
'q' = "0xB0",
'r' = "0xB1",
's' = "0xB2",
't' = "0xB3",
'u' = "0xB4",
'v' = "0xB5",
'w' = "0xB6",
'x' = "0xB7",
'y' = "0xB8",
'z' = "0xB9",
'-' = "0xE3",
'?' = "0xE6",
'!' = "0xE7",
'.' = "0xE8",
',' = "0xF4",
'0' = "0xF6",
'1' = "0xF7",
'2' = "0xF8",
'3' = "0xF9",
'4' = "0xFA",
'5' = "0xFB",
'6' = "0xFC",
'7' = "0xFD",
'8' = "0xFE",
'9' = "0xFF"
```

EstebanFuentealba, 2023-04-23:
And for moves, you can use this dictionary as a reference.

```bash
"Nothing" = "0x00",
"Pound" = "0x01",
"Karate Chop" = "0x02",
"Double Slap" = "0x03",
"Comet Punch" = "0x04",
"Mega Punch" = "0x05",
"Pay Day" = "0x06",
"Fire Punch" = "0x07",
"Ice Punch" = "0x08",
"Thunder Punch " = "0x09",
"Gust" = "0x10",
"Wing Attack" = "0x11",
"Whirlwind" = "0x12",
"Fly" = "0x13",
"Bind" = "0x14",
"Slam" = "0x15",
"Vine Whip" = "0x16",
"Stomp" = "0x17",
"Double Kick" = "0x18",
"Mega KIck" = "0x19",
"Jump KIck" = "0x1A",
"RollingKick" = "0x1B",
"Sand Attack" = "0x1C",
"Headbutt" = "0x1D",
"Horn Attack" = "0x1E",
"Fury Attack" = "0x1F",
"Horn Drill" = "0x20",
"Tackle" = "0x21",
"Bodyslam" = "0x22",
"Wrap" = "0x23",
"Take Down " = "0x24",
"Trash" = "0x25",
"Double Edge" = "0x26",
"Tail Whip" = "0x27",
"Poison Sting" = "0x28",
"Twin Needle" = "0x29",
"Pin Missle" = "0x2A",
"Leer" = "0x2B",
"Bite" = "0x2C",
"Growl" = "0x2D",
"Roar" = "0x2E",
"SuperSonic" = "0x2F",
"Sonic Boom" = "0x30",
"Disable" = "0x32",
"Acid" = "0x33",
"Ember" = "0x34",
"FlameThrower" = "0x35",
"Mist" = "0x36",
"Water Gun" = "0x37",
"Hydro Pump" = "0x38",
"Surf" = "0x39",
"Ice Beam" = "0x3A",
"Blizzard" = "0x3B",
"PsyBeam" = "0x3C",
"Bubble Beam" = "0x3D",
"Aurora Beam" = "0x3E",
"Hyper Beam" = "0x3F",
"Peck" = "0x40",
"Drill Peck" = "0x41",
"Submission" = "0x42",
"Low Kick" = "0x43",
"Counter" = "0x44",
"Sesmic Toss" = "0x45",
"Strength" = "0x46",
"Absorb" = "0x47",
"Mega Drain" = "0x48",
"Leech Seed" = "0x49",
"Growth" = "0x4A",
"Razor Wind" = "0x4B",
"Solar Beam" = "0x4C",
"Poison Powder" = "0x4D",
"Stun Spore" = "0x4E",
"Sleep Powder" = "0x4F",
"Petal Dance" = "0x50",
"String Shot" = "0x51",
"Dragon Rage" = "0x52",
"Fire Spin" = "0x53",
"ThunderShock" = "0x54",
"ThunderBolt" = "0x55",
"ThunderWave" = "0x56",
"Thunder" = "0x57",
"Rock Throw" = "0x58",
"EarthQuake" = "0x59",
"Fissure" = "0x5A",
"Dig" = "0x5B",
"Toxic" = "0x5C",
"Confusion" = "0x5D",
"Psychic" = "0x5E",
"Hypnosis" = "0x5F",
"Meditate" = "0x60",
"Agility" = "0x61",
"Quick Attack" = "0x62",
"Rage" = "0x63",
"Teleport" = "0x64",
"Night Shade" = "0x65",
"Mimic" = "0x66",
"Screech" = "0x67",
"Double Team" = "0x68",
"Recover" = "0x69",
"Harden" = "0x6A",
"Minimized" = "0x6B",
"Smoke Screen" = "0x6C",
"Confuse Ray" = "0x6D",
"WithDraw" = "0x6E",
"Defence Curl" = "0x6F",
"Barrier" = "0x70",
"Light screen" = "0x71",
"Haze" = "0x72",
"Reflect" = "0x73",
"Focus" = "0x74",
"Bide" = "0x75",
"Metronome" = "0x76",
"Mirror Move" = "0x77",
"Self Destruct" = "0x78",
"Egg Bomb" = "0x79",
"Lick" = "0x7A",
"Smog" = "0x7B",
"Sludge" = "0x7C",
"Bone Club" = "0x7D",
"Fire Blast" = "0x7E",
"Waterfall" = "0x7F",
"Clamp" = "0x80",
"Swift" = "0x81",
"Skull Bash" = "0x82",
"Spike Cannon" = "0x83",
"Constrict" = "0x84",
"Amnesia" = "0x85",
"Kinesis" = "0x86",
"Softboiled" = "0x87",
"Hi-Jump Kick" = "0x88",
"Glare" = "0x89",
"Dream Eater" = "0x8A",
"Poison Gas" = "0x8B",
"Barrage" = "0x8C",
"Leech Life" = "0x8D",
"Lovely Kiss" = "0x8E",
"Sky Attack" = "0x8F",
"Transform" = "0x90",
"Bubble" = "0x91",
"Dizzy Punch" = "0x92",
"Spore" = "0x93",
"Flash" = "0x94",
"Psywave" = "0x95",
"Splash" = "0x96",
"Acid Armor" = "0x97",
"Crabhammer" = "0x98",
"Explosion" = "0x99",
"Fury Swipe" = "0x9A",
"Bonemerang" = "0x9B",
"Rest" = "0x9C",
"Rock Slide" = "0x9D",
"Hyper Fang" = "0x9E",
"Sharpen" = "0x9F",
"Conversion" = "0xA0",
"Tri Attack" = "0xA1",
"Super Fang" = "0xA2",
"Slash" = "0xA3",
"Substitute" = "0xA4",
"Struggle" = "0xA5"
```

EstebanFuentealba, 2023-04-23:
I hope the information is helpful, and if you can send a PR with the functionality, it would be great.

LuxStatera, 2023-04-23:
Thank you so much for the information! Unfortunately I'm not able to make sense of it. I see in the pokemon_data.h there is some code under the TRAINER NAME that I thought might spell out Flipper or the Trainer ID number. I tried to use that to test out if I could match the code to the info you provided but I can't seem to find the matching info or maybe that code is something else. I am still a bit unfamiliar with GitHub and not sure what a PR is used for. My apologies I'm a total beginner.. Once I understand it then I would like to type out each Pokémon with their own moves and stats to submit. Maybe the app could be adjusted so the traded Pokémon get the same Trainer code as the cartridge so there won't be any issues with not having the correct badges for the high levels

LuxStatera, 2023-04-23:
After looking over it a bit more I realized the TRAINER NAME does spell out Flipper. For some reason I was looking at the capitalized letters which was why they were not matching up. My bad. I changed the TRAINER NAME to the same one I created in my new save file on my cartridge. I will test to see if that allows the traded Pokemon to listen without badges. The pokemon_data.h file has only Mew listed with 4 different attacks that all traded Pokemon will also have but I noticed that their type is different for each one. Where is it getting that data if it is not listed in the pokemon_data.h file? I can add all 151 with custom attacks and specs once I understand it a little more

LuxStatera, 2023-04-23:
After changing the name to the same one on my cartridge I made the trade and battled a wild Pokemon but it wouldn't listen. Most likely it has to do with the ID number but I don't see how to change that. The only levels listed in the pokemon_data.h file are in the "6 pokemon" section but after changing all of them to the hexadecimal equivalent of 15 it still showed the default 74 after a trade. I am unclear as to what the 6 pokemon, 6 trainer names, and 6 nicknames sections are for

LuxStatera, 2023-05-11:
I've made changes to pokemon_data.h and regenerated the .fap but nothing changes with the traded Pokémon besides trainer name. How do I edit the individual Pokémon? In the pokemon_data.h file there is a section with only "MEW" but I'm not seeing where that fits in because the edits aren't recognized

R4g3D, 2023-09-11:
I should have looked here while I was working on the code 😂 I manually scrapped a lot of the data you had here already 🤦🏻

@EstebanFuentealba you can close this issue since it's now part of the app.

closed 2023-09-27 by EstebanFuentealba