kbembedded/Flipper-Zero-Game-Boy-Pokemon-Trading #9: Furi crash when selecting pokemon

matthewwwillard, 2023-04-28:
### Describe the bug.

Flipper crashes on a furi_check failed after selecting a pokemon. When building from source, looked to be an issue when opening up the serial / gpio ports.

### Reproduction

1. Open app
2. Select Pokemon

### Target Framework

latest official firmware

EstebanFuentealba, 2023-04-28:
Does it show any error? or simply goes back to the Flipper Zero's home screen? 0.82.3 ?

matthewwwillard, 2023-04-28:
[image]
Yea, 0.82.3

kbembedded, 2023-05-10:
I've also had the same issue with latest binary pulled from flipC.org. Havn't tried hand-building yet. Spinning up on some hardware stuff to make a board before I really dive in to this.

EstebanFuentealba, 2023-05-10:
this week or maybe the other I get a zero flipper to try

kbembedded, 2023-05-17:
I did some quick debug rounds (I'm still learning all of the workflows) and found this patch so far at least fixes the `furi_check` crash and gets to a point where it wants to connect to a real Game Boy:

```
diff --git a/pokemon_app.cpp b/pokemon_app.cpp
index ce274d6..2ef73e9 100644
--- a/pokemon_app.cpp
+++ b/pokemon_app.cpp
@@ -124,10 +124,7 @@ extern "C" int32_t pokemon_app(void* p) {
     furi_hal_light_set(LightRed, 0x00);
     furi_hal_light_set(LightGreen, 0x00);
     furi_hal_light_set(LightBlue, 0x00);
-    view_dispatcher_enable_queue(app->view_dispatcher);
-    view_dispatcher_attach_to_gui(app->view_dispatcher, app->gui, ViewDispatcherTypeFullscreen);
     //switch view  and run dispatcher
-    view_dispatcher_switch_to_view(app->view_dispatcher, AppViewSelectPokemon);
     view_dispatcher_run(app->view_dispatcher);
 
     // Free resources
@@ -135,4 +132,4 @@ extern "C" int32_t pokemon_app(void* p) {
     furi_record_close(RECORD_GUI);
 
     return 0;
-}
\ No newline at end of file
+}
diff --git a/views/trade.cpp b/views/trade.cpp
index ad503bb..f6d1f93 100644
--- a/views/trade.cpp
+++ b/views/trade.cpp
@@ -327,6 +327,7 @@ void trade_enter_callback(void* context) {
         GpioModeInterruptRise,
         GpioPullNo,
         GpioSpeedVeryHigh); // <-- This line causes the "OK" to stop functioning when exiting the application, so a reboot of the Flipper Zero is required.
+    furi_hal_gpio_remove_int_callback(&GAME_BOY_CLK);
     furi_hal_gpio_add_int_callback(&GAME_BOY_CLK, input_clk_gameboy, trade);
 
     // furi_hal_gpio_disable_int_callback(&GAME_BOY_CLK);
@@ -384,4 +385,4 @@ void trade_free(App* app) {
 View* trade_get_view(App* app) {
     furi_assert(app);
     return app->trade->view;
-}
\ No newline at end of file
+}
```

The first calls that were removed from `pokemon_app()` were being called prior in `pokemon_alloc()` and calling a second time will fail but only in builds I did fully from source. Not sure why that doesn't trip firmware, likely a debug/release build thing.

The added call to `remove_init_callback` was needed since that pin already had a callback set up on it to `nfc_isr()`. This might be a bug in flipper FW, but I'm not sure.

Hope this helps. I'll be playing around with this more when I have more time.

kbembedded, 2023-05-20:
After looking at this a bit closer, I do really think this is a bug with the FW.

The reason the issue appeared is that there was a furi_assert that was changed to a runtime furi_check. So, this has always technically been a problem.

I've posted some notes in the Flipper discord and am waiting on feedback to see if its worth opening a GH issue about.

kbembedded, 2023-05-23:
This is a known issue, not marked in current GPIO/documentation, and is not safe _at this time_ to be used as EXTI for any other application. Sounds like it is being addressed in the refactor and will potentially be safe to use.

https://github.com/flipperdevices/flipperzero-firmware/issues/2690

Potential options moving forward:
- Wait. Right now the #11 draft has a workaround for this, but it likely means NFC won't work when you exit the application and need to restart flipper (I can try seeing if the context of a FAP can understand and correctly set back up the `nfc_isr` callback). It is also unknown how long the refactor will take.
- Move the interface to different pins. I'm currently working with the Mayhem board designer to implement a daughter board that is a game boy link socket in order to add game boy printer compatibility to it for it to print photos taken by the camera directly from the Flipper. I was also planning on putting together a standalone PCB just for link cable connectivity. It would be nice if this tool and any other tools were all on the same page about what pins to use. This would also eliminate the need of the workaround for the callback and would likely be immediately compatible with future FW releases. However, much of the documentation here would need to change.

EstebanFuentealba, 2023-05-24:
Yes, is not safe at this time to be used as EXTI. PC3 uses EXTI3 line which is shared among all Px3 pins on all ports.  https://github.com/flipperdevices/flipperzero-firmware/issues/2538 

Moving the interface to different pins is the best option, although it would require updating the documentation, images, and wiring diagrams.

PR #12 (2023-06-11): Fix furi crash
Fixes #9 

Also cleans up some double function calls.

While @EstebanFuentealba and I briefly chatted about moving pins due to conflicts with Flipper firmware; this can still work (just need to reboot the Flipper Zero before using NFC again), and once the API settles in OFW 1.0 that can be a choice made then. For now, it doesn't make sense to re-do all of the photos and graphics in this repo.
closed 2023-06-11 by EstebanFuentealba