kbembedded/Flipper-Zero-Game-Boy-Pokemon-Trading #10: Question about connection problems

FabiFNA, 2023-05-06:
Neither the gameboy nor the Flipper recognise the connection. I've connected seemingly everything accordingly. If I'm wrong please correct me by the following pictures:
[image]

This is how it looks when it's attached to the Flipper:
[image]

Just to clear things up, I've used a multimeter to check if everything is connected correctly and judging by the multimeter it is. None of the channels of the cable are connected to multiple GPIO ports. I've tried out an older version of the Flipper App and the newest release, neither worked. Also, using the 5V doesn't make any difference. Any ideas?

EstebanFuentealba, 2023-05-06:
[image]

Do not connect the 5v because you could kill the Gameboy or the flipper.

FabiFNA, 2023-05-06:
Thank you very much that did the trick.

closed 2023-05-06 by FabiFNA