kbembedded/Flipper-Zero-Game-Boy-Pokemon-Trading #20: [Pokémon Yellow] Trading always gives glitched Rhydon

theblackhole, 2023-11-04:
### Describe the bug.

Trading App version: 1.3
Flipper Firmware: 0.94.1
Game: Pokémon Yellow, French version

I can't trade: it forces me to receive a glitched Rhydon, instead of the Pokémon I've want.

This Rhydon is glitched: on the first trade I made, I couldn't open my Pokémon list anymore (white screen and couldn't quit) after the broken trade.

The video below shows the second trade where I am forced to exchange this Rhydon which breaks my save for a Rhydon which doesn't break my save but is still glitched (no name, strange stats, broken icons)

Note: I tried with and without resistors on CLK, the issue is the same.

### Reproduction

Video of the issue (Rhynoferos = Rhydon and Bulbizarre = Bulbasaur, the Pokémon I chose in the app): https://youtu.be/sYfjd7b4AdA

1. Choose a Pokémon to trade on the flipper
2. Start a trading session
3. On the game's trade screen, the flipper has 2 visible Pokémons: a (glitched) Rhydon first and, below it, the Pokémon I've chosen
4. Flipper always select this Rhydon to trade so you will receive a glitched Rhydon no matter what.

kbembedded, 2023-11-04:
Interesting.

Does this happen no matter what Pokemon and stats you select?
Are you using just the defaults for Bulbasaur?
Do you only have Pokemon Yellow or do you have Red/Blue as well?

I've only ever done testing on USA Blue, but, I think @EstebanFuentealba did most of the work against Yellow (not sure exactly what region) and hasn't run in to issues as far as I am aware.

I've been working on an overhaul of the Trade routines, cleaning up the areas where it is fragile. Are you able to build and test from my WIP branch and let me know if it continues? https://github.com/kbembedded/Flipper-Zero-Game-Boy-Pokemon-Trading/tree/trade-features

For what its worth, I don't use the series resistor, and the way the code is set up it should never be an issue (that is, the Flipper side will never try and drive the clock pin).

theblackhole, 2023-11-05:
First, there is a detail that I just thought of and that could be a lead: my save/Pokémon trainer's name has a one of the Pokémon special characters, it's "☥ANUBIS" (Well in reality it's"♀ANUBIS" but with so few pixels, there is no difference)

> Does this happen no matter what Pokemon and stats you select?

Yup, my first trade attempt was an overpowered Mew, with custom name and a Peck attack

> Are you using just the defaults for Bulbasaur?

Yep, for this second attempt, all defaults

> Do you only have Pokemon Yellow or do you have Red/Blue as well?

Unfortunately no, besides Yellow, I only have the next gen atm (Gold)

> Are you able to build and test from my WIP branch and let me know if it continues? https://github.com/kbembedded/Flipper-Zero-Game-Boy-Pokemon-Trading/tree/trade-features

Sure I'll try

> For what its worth, I don't use the series resistor, and the way the code is set up it should never be an issue (that is, the Flipper side will never try and drive the clock pin).

Got it

Btw combats with this glitched Rhydon are quite fun to watch: HP bars extend beyond limit, text doesn't mean anything, it even corrupts Pikachu and people 😁
[Imgur Gallery](https://imgur.com/a/grP0yr7)

kbembedded, 2023-11-05:
> First, there is a detail that I just thought of and that could be a lead: my save/Pokémon trainer's name has a one of the Pokémon special characters, it's "☥ANUBIS" (Well in reality it's"♀ANUBIS" but with so few pixels, there is no difference)

You might be on to something. I took closer look at your video. The Flipper's trainer name has ♀ prepended to it.

I think this is something I recently fixed and my WIP branch should work for you.

tl;dr: (I can go in to more detail on the issue if we confirm that my WIP has any effect on the bug) the code to go from synchronizing RNG between the linked units to start swapping real data between them very loosely matches control bytes; the ♀ symbol would match this and the Flipper might be thinking its part of the preamble still.

Looking at the data struct, if the data the Flipper sent was shifted by one, that would push the 0x01 `party_count` value in to the first byte of the `party_member`s. 0x01 is the hex value of a Rhydon. The rest of the data just kind of goes off the rails.

> Btw combats with this glitched Rhydon are quite fun to watch: HP bars extend beyond limit, text doesn't mean anything, it even corrupts Pikachu and people 😁 [Imgur Gallery](https://imgur.com/a/grP0yr7)

Those are fantastic. I ran in to a bunch of hilarious effects when doing work on this a while ago. One of them actually crashed my game if I opened Start->Pokemon. I eventually figured out that the HP value was so large, the game would try to render an HP bar that long, and it basically clobbered the whole screen before just hard-locking the system. I luckily was able to box it up and release it and my game no longer had issues.

This was also a fun one:

[image]

theblackhole, 2023-11-05:
> I think this is something I recently fixed and my WIP branch should work for you.

I confirm. I built your fork, and it fixed my issue (after removing the resistors because it wasn't detecting the flipper). No more Rhydon, no more "♀Flipper", only the Pokémon I've chosen.
To be sure, I exchanged my glitched Rhydon for the default Bulbazaur
Then I catched and named a Pokémon with a lot of special chars and I exchanged it for an overpowered weird MewTwo (Fire type and with Peck attack).

Everything worked as expected, nice work! :tada:

> tl;dr: <details><summary>(...)</summary>(I can go in to more detail on the issue if we confirm that my WIP has any effect on the bug) the code to go from synchronizing RNG between the linked units to start swapping real data between them very loosely matches control bytes; the ♀ symbol would match this and the Flipper might be thinking its part of the preamble still. If that is the case, that means any trainer names with a number, or a couple other specific symbols, as the first character would also suffer from this. If you had that symbol in any other character position, I don't think it would be a problem on the latest release.
> 
> Looking at the data struct, if the data the Flipper sent was shifted by one, that would push the 0x01 `party_count` value in to the first byte of the `party_member`s. 0x01 is the hex value of a Rhydon.</details> The rest of the data just kind of goes off the rails.

Thanks for the explaination, it makes sense.

> This was also a fun one:
><details>
>  <summary>Image</summary>
>[image]
></details>

lmao this reminds me of the Sesame Street "8 to infinity" meme

[image]

kbembedded, 2023-11-05:
> I confirm. I built your fork, and it fixed my issue (after removing the resistors because it wasn't detecting the flipper). No more Rhydon, no more "♀Flipper", only the Pokémon I've chosen. To be [...]

:tada: 

Glad to hear it.

@EstebanFuentealba keep this issue open for now for anyone else running in to it. I'll mark it as being resolved by a new PR once I open it.

> ... exchanged it for an overpowered weird MewTwo (Fire type and with Peck attack).

An interesting note on types if you were not aware. The Flipper can set them to whatever, but if you view the Pokemon's stats, it will always show the Pokemon's defaults. The save data keeps a writable copy of types that is normally only ever updated when the Pokemon evolves or uses an attack that could temporarily change its type, Transform, for example. Since Mewtwo doesn't evolve, as long as you never use an attack with a type change effect, it will stay as Fire/Peck (I assume Peck is Flying?)

Before I understood that quirk of types, I created a Ghost/Fire Mew and was baffled when it was unaffected by physical attacks lol.

> lmao this reminds me of the Sesame Street "8 to infinity" meme
> 
> [image]

:rofl:

theblackhole, 2023-11-05:
> it will stay as Fire/Peck (I assume Peck is Flying?)

The Mewtwo I made is a Psy/Fire type (it only shows Type 1: Psy as espected) and I replaced one of the attacks with Peck

kbembedded, 2023-11-06:
I also just learned that, Trainer and Pokemon names cannot contain numbers. It just so happens a few symbols end up with their upper nibble being hex F. So there are some other tweaks I also need to make. In theory though, the gameboy should render them, but, it will cause issues with the patch list since that does not cover names. I always wondered why, and its because they should never have the byte that the patch list patches.

PR #21 (2023-11-08): Trade: Overhaul for features and fixes. Bump to 1.4
See the commit logs, most of the detail is there, but this essentially:
- Improves trade logic handling (which happened to fix #20)
- Adds patch list support bidirectionally (which fixes #18)
- Flipper receives traded Pokemon, and can modify Pokemon, and can easily trade back and forth (fixes #19)
- Isolates Trade's scope
- Simplifies trade/link state handling
- Tries to document the trade protocol
- Cleans up TRADING animations
- Reduces time spend in ISR by enabling interrupt on both edges and handling data clocking there
- No longer allows [... cut]
commit c617ce38 (2023-11-08): trade: Fixes, features, and improvements

This commit wraps up a number of changes that aim to improve the trade
process, its robustness, add some features, and fix some bugs.

Change list in rough order of importance:

More robust logic during trading which more correctly syncs states
between the Flipper and Game Boy.
Fixes #20 as a side effect.

Implement trade patch list. This allows for the Flipper to both send and
receive patch data.
Fixes #18

Ability for the Flipper to correctly receive the Pokemon traded to it
and put that Pokemon in the Flipper's trade buffer. This allows for
trading  [... cut]
closed 2023-11-08 by kbembedded