kbembedded/Flipper-Zero-Game-Boy-Pokemon-Trading #28: Update the Pokemon Trading app for the SDK version 0.100 f7

furippa, 2024-04-27:
Hello!

Your Pokemon Trading app version `1.2` fails to build with the SDK version `0.100.1-rc f7`. Please update the app in your repository and create a new pull request to the catalog repository with the updated manifest file.

> IMPORTANT: Don’t forget to update the app version and commit version in the manifest file.

[View logs](http://192.168.1.38:8000/application/64971d4d577d519190ede5c6/page)

Flipper Team

furippa, 2024-04-27:
I am again here!

Your Pokemon Trading app version `1.2` fails to build with the SDK version `0.100.1-rc f7`. Please update the app in your repository and create a new pull request to the catalog repository with the updated manifest file.

> IMPORTANT: Don’t forget to update the app version and commit version in the manifest file.

[View logs](http://192.168.1.38:8000/application/64971d4d577d519190ede5c6/page)

Flipper Team

closed 2024-04-27 by furippa

limness, 2024-04-27:
Hello!

We apologize. The issue was sent by mistake.

Flipper Team