kbembedded/Flipper-Zero-Game-Boy-Pokemon-Trading #29: Feat: Gen II IR Mystery Gift

kbembedded, 2024-05-07:
### Describe the process.

Add support to link between the Flipper and a Gen II Pokemon game over IR for Mystery Gift (Split from #24).

I've only briefly looked over the IR system, but, it doesn't look straightforward to implement. The API is currently only set up for TX xor RX and not easily able to do bidirectional (note: this is not IrDA, this is generic 38 khz IR that sends data back and forth). It might be possible to hack together something at the application level, however I think this would greatly benefit from in-firmware support, alongside a runtime "custom" protocol decoder implementation that can pack and unpack data directly.

### Game boy

Color

jaytea33, 2024-06-03:
Just in case no one stumbled upon this yet and wants to take a crack at it, this may be useful

https://projectpokemon.org/home/forums/topic/43930-mystery-gift-reverse-engineering-of-ir-protocol/

kbembedded, 2024-06-03:
Thanks!

I'll add that to my list of resources for when I get around to implementing this.

kbembedded, 2026-09-20:
> I've got it working with a custom GPIO board!! Please see [my answer](https://projectpokemon.org/home/forums/topic/43930-mystery-gift-reverse-engineering-of-ir-protocol/page/2/#findComment-300951) in the forum post that was shared above. I'll clean up everything and if anyone is interested, I can share the code and diagram for the parts. It's a bit tidious to align properly but I hope to fix that with the final version with maybe a 3d printed case to help with alignement.

Apologies for the delay in responding.

That is awesome to see! Congrats!

Now that its been a bit of time, how has the project progressed? I've been considering picking this back up and it looks like the actual GBC schematic has some clever auto-balancing circuit to handle ambient IR light. I've been thinking on a new EXT interface board that has actual support for GBA (which can use UART and higher speed SPI and needs the real SPI interface on the flipper, so a different pinout) and it may be possible to add this circuit to it and fold in mystery gift to the main application.

If you want to share your circuit and/or put together a PR for this repo that includes that code as well as a schematic I'd be happy to consider that.

TheRealNox, 2026-10-08:
> Apologies for the delay in responding.
>
> That is awesome to see! Congrats!
>
> Now that its been a bit of time, how has the project progressed? I've been considering picking this back up and it looks [...]

Now it's my turn to be late to reply haha
I've pushed all my code in here: https://github.com/TheRealNox/flipper-pokemon-mystery-gift, I haven't had the time to cleanup my POC, so it's not even solder on the proper board yet. I'll try to find some times to at least create a proper schema as right now, I only have a picture of the monstruosity of a board I created haha.