kbembedded/Flipper-Zero-Game-Boy-Pokemon-Trading #33: Feat: Pokemon Emerald secret word generator

kbembedded, 2024-05-10:
### Describe the process.

(This was proposed to me in Discord, I don't have much knowledge of the subject, but, could be a neat tool to have in place)

An implementation of this: https://pokemonexperte.de/smaragd/geheimwort.php?language=en

EstebanFuentealba, 2024-05-11:
I would place all the new ideas/enhancements (IR Mystery Gift, Pokemon Trading Card Game Card Pop, Secret Word Generator) into different and distinct applications from trading for several reasons:

- It simplifies the experience for the end user by reducing the number of options/menus to navigate.
- It's easier for developers to contribute via PRs to smaller projects, which are easier to comprehend and maintain.
- Each project maintains a singular purpose and context.
- Easier to maintain the code.

kbembedded, 2024-05-11:
> I would place all the new ideas/enhancements (IR Mystery Gift #29, Pokemon Trading Card Game Card Pop #30, Secret Word Generator #33 , time changing #32) into different and distinct applications from trading for several reasons:

Normally, I would agree with you and have taken that stance before when we have talked about it. However, I disagree with that concept for these specific features.

> 
>     * It simplifies the experience for the end user by reducing the number of options/menus to navigate.
>     * Each project maintains a singular purpose and context.

With the ecosystem of the Flipper, I have found that people tend to load up anything and everything they can get their hands on (IRDB, subghz DB, every application in the app catalog, etc.). I would argue that there would be more menu navigation to get to each individual application than there would be accessing very closely related tools all in the same app. There are currently 40+ applications in the GPIO section in the official app catalog, and having even a small fraction of those installed would require just as much navigation to get to each of your proposed 5 different apps compared to just scrolling down a few more entries on the Trade app.

The other argument I have is that with the MALVEKE ecosystem of tools, they are all named prefixed with "GAME BOY" in the Flipper, this takes up half of the visible filename in the folder listing. e.g. the trade tool is "GAME BOY Pokemon Tr..." Having only a few characters to describe the application at a glance without having to stop on it and wait for the text to scroll to see the full name is very limiting.

Because of these, adding smaller, simple tools that all have to do with Game Boy Pokemon trading would indeed make the user experience more streamlined.

> 
>     * It's easier for developers to contribute via PRs to smaller projects, which are easier to comprehend and maintain.
>     * Easier to maintain the code.

I will concede this point, the codebase is not the best right now. I am planning on a refactor for the menuing system to make it more like in-firmware menuing is set up, allowing each menu segment to be a bit more isolated and self contained.

However, adding these features at this moment would only require understanding of the main menuing system. There would be more complexity in figuring out how to properly represent all of the characters that one could have as part of their trainer name with the limited keyboard input.

But at the end of the day, it's your repo. Feel free to close those issues as wontfix, or don't approve the merge requests if they ever come through if you don't think they belong here.