kbembedded/Flipper-Zero-Game-Boy-Pokemon-Trading #40: All sprites are missing. Replaced with "?"

LuxStatera, 2024-08-17:
### Describe the bug.

I just updated RM firmware with the updated Pokémon Trading app so I could test out the Gen 2 trading. When I choose the Pokémon selection option all the sprites are missing. I went through them and they show the names and numbers but the sprites are replaced with a  question mark in a circle

### Reproduction

1. select Pokémon Trading app
2. choose gen 1 or gen 2
3. select Pokémon
4. sees the first option which is Bulbasaur. Sprite is missing and replaced with a question mark
5. click right to scroll through Pokémon and they all have a question mark

kbembedded, 2024-08-17:
Interesting bug.

The sprites are stored as a FAP asset; which are bundled up with the .fap, and at application load the OS ensures the asset is unpacked and matches checksum, if either fail its unpacked just before application launch. If there is an issue with the SD card, asset unpacking in RM, or asset path macros in RM, those could all cause issues. The sprites are kept as an asset and loaded on-demand; 251 56x56 px sprites take up a lot of memory otherwise.

The round bubble with the `?` is the MissingNo. icon from later generations and is kept in memory. If sprite loading fails for any reason, that sprite is returned and used instead.

I'll be honest and say that how sprite information is loaded is probably a bit non-ideal, but, there should be no issues with it. Works in 0.105.0 official release (I havn't tested against dev, which RM may be pulling from, I'm not sure).

Can you open `log debug` through flipper console and copy/paste that here? I might be able to figure something out from that.

Does the trade still work even with the sprites missing? If not, then I'll try and look in to it.

Otherwise, if its just the cosmetics of the sprites then addressing this is going to be pretty low priority. I generally only develop against official, as keeping track of all of the customs, where they are in the API, any changes they might have, etc., doesn't really interest me and becomes time consuming to test for issues like this. If someone makes a PR for this, or the bug appears in official firmware at some point, I'll address it further then.

kbembedded, 2024-08-18:
Just had a moment to test against latest official dev (99655c15e454ffdbd4bada76bd45fa41fd28f7c1) and I'm not able to reproduce it there.

This is definitely something specific to RM. I'm interested in console logs still if you're able to grab them, they might be able to tell me something that could be a quick fix. But otherwise, if trades still work, this is a low priority issue for me to look in to.

kbembedded, 2024-08-27:
Had a moment and tested this against `RM0825-0001-0.104.1-ff3298a` and was unable to reproduce.

closed 2024-08-27 by kbembedded