kbembedded/Flipper-Zero-Game-Boy-Pokemon-Trading #43: Crashes when i try to make a trade

fbellott, 2025-02-25:
### Describe the bug.

Each time i try to make a trade, the Flipper crashes with the message you can see on the screenshot.
I have a proof that it worked for me on 2024/09/07, so on the Flipper firmware [0.105.0](https://github.com/flipperdevices/flipperzero-firmware/releases/tag/0.105.0) at maximum.
Tried with Pokemon Yellow and Cristal, both crash
Tried with 1.1.2 and 1.2.0 firmwares, both crash
I would try with 0.105.0, but they recommend to not downgrade from 1.1.2

[image]

### Reproduction

Just perform a trade.

### Logs

```Text

```

kbembedded, 2025-02-26:
I'm unable to reproduce this. Tested on official flipper firmware 1.2.0 against Crystal, Yellow, Gold, and Blue I was able to trade a pokemon back and forth without issue.

This leads me to believe its something unique to your hardware or your party. Can you please let me know the following?

- What GB device you are using
- What link cable you are using
- Are you using a real cartridge or a ROM cart like Everdrive
- What is the pinout you are using to the flipper and if you are using an off the shelf adapter (Flipper-GB-Link, MALVEKE, etc.)
- What is the name of your character in the game(s)
- What are the names of the pokemon in your party.
- At what point exactly do you see the crash happen (e.g. before you enter the trade room, after pressing A on the trade table, after selecting a pokemon and pressing Trade, after the trade animation completes, etc.)

belotte55, 2025-02-26:
Thank you for your answer, your interest on my issue, and for this tool 🤗

OK I found it !!

So im playing on a Analogue Pocket, with [this core](https://github.com/budude2/openfpga-GBC). And the crash appears when I try to make the trade in "Fastforward mode". I just tried in normal mode, and it works ! While it was a 100% crash in fastforward

I keep the answers of your questions ⬇ 
---
First to know, I already trade some Pokemons few months ago (in September)

- What GB device you are using
  An Analogue Pocket (already worked with it)
- What link cable you are using
  A cheap Aliexpress one (already worked)
- Are you using a real cartridge or a ROM cart like Everdrive
  I use a ROM loaded by a Core, on the Pocket (already worked)
- What is the pinout you are using to the flipper and if you are using an off the shelf adapter (Flipper-GB-Link, MALVEKE, etc.)
  A DIY one. I can't remember where I found the schema. It integrates a LED. (already worked)
- What is the name of your character in the game(s)
  He's called "SACHA". Why is this important ? 🤔 
- What are the names of the pokemon in your party.
  The default names (i NEVER change them !) but in french: Kadabra, Aligatueur, Mystherbe, Togepi, Lippouti and maybe some others on the PC (already worked in french)
- At what point exactly do you see the crash happen (e.g. before you enter the trade room, after pressing A on the trade table, after selecting a pokemon and pressing Trade, after the trade animation completes, etc.)

kbembedded, 2025-02-26:
> So im playing on a Analogue Pocket, with [this core](https://github.com/budude2/openfpga-GBC). And the crash appears when I try to make the trade in "Fastforward mode". I just tried in normal mode, and it works ! While it was a 100% crash in fastforward

Ah, yes, that could do it. I'm not exactly sure what fastforward does (I wasn't able to find any details on specifically how it works) but I would assume it runs the clock faster which can likely affect transfer speed and synchronization. I don't have an AP to test with, so I cannot be completely certain.

Thanks for following up on this. Glad you are able to get it to work. Let me know if you run in to any other issues.

>     * What is the name of your character in the game(s)
>       He's called "SACHA". Why is this important ? 🤔

While it shouldn't be an issue, there has been an issue in the past with characters in names that are valid, but trip up the Flipper application. e.g. #20

kbembedded, 2025-02-27:
Many thanks, friend!

fbellott, 2025-02-27:
Thanks to you for your work ! 🤩